MakeYourPet/hexapod #1: Please add a README.md

johnsabilla, 2022-10-09:
Please add a README.md ty ty

moebis, 2022-10-09:
Missing parts list too. BTW, thanks for this awesome project. Really excited to start on it.

ARobBot, 2022-10-17:
A bill of materials would be greatly appreciated. Ta

SupMap, 2022-10-18:
Awesome project!

A parts list would be a very great addition, also maybe releasing the apps (software) as open source so others can contribute?

Thanks!

johnsabilla, 2022-10-18:
all of these are just .STL files so just the 3D printable parts. It doesn't include the other electronic hardware.

Beenay25, 2022-10-24:
I'd just be happy knowing what size the servos are. 😁

moebis, 2022-10-24:
I think this project might have died before it took off. Just radio silence. Too bad, I kinda thought it was promising.

MakeYourPet, 2022-10-24:
> 

README and LICENSE files are coming, but so many people are already building this hexapod and didn't need those to get started, and unfortunately I have other priorities.

On a separate note, comments like this are one of the main reasons that I am not open-sourcing my code.

Beenay25, 2022-10-24:
> > I think this project might have died before it took off. Just radio silence. Too bad, I kinda thought it was promising.
> 
> README and LICENSE files are coming, but so many people are already building this hexapod and didn't need those to get started, and unfortunately I have other priorities.
> 
> On a separate note, comments like this are one of the main reasons that I am not open-sourcing my code.

We do appreciate all your work on the project. It's exciting!
I can imagine many people have been able to make their own with little instruction. Some of us are more noobs and might need a bit more direction.

I'll get there eventually, but I'm really looking forward to getting specs and parts lists to help. 😁

MakeYourPet, 2022-10-24:
I understand, and appreciate your interest in this project as well. Then I would suggest for you to hold off a month or two, until we resolve all the issues and refine the hard edges while some others finish building their hexapods. Because in the current format you would need to be able to pick your electronics and finish the wiring on your own, and I don't have a detailed step by step assembly instructions ready yet.

johnsabilla, 2022-10-24:
Just join their discord. Lots of people are ahead of building the pod

MakeYourPet, 2022-10-25:
Go to www.makeyourpet.com and click on the Discord button to join. Thank you.

MakeYourPet, 2022-10-25:
I also added some of the useful links to the README file. I was under the impression that all of the folks here know about my YouTube channel and my Discord server. Now your comments make sense. I hope you build and enjoy the hexapod.

MakeYourPet, 2022-10-25:
And just added a fan-made wiring diagram and two separate fan-made parts lists.

Beenay25, 2022-11-08:
This is all looking so good now! Exciting!

Thank you so much for all the updates.