MakeYourPet/hexapod #24: Can you use 360° servo instead of 180°?

Snoozer-94, 2024-06-07:
Hey, yh the title pretty much asks the question. Looking into buying some servos and found that the 360 are cheaper than the 180, was wondering if that would be an issue at all. Have experience with electronics and some coding but when it comes to robotics I'm pretty new.

Snoozer-94, 2024-06-10:
Pretty much answered my own question through research, for anyone wondering basically 360 servos (aka continuous servos) don't use the values in the code the same way 180 servos do. Instead of the servo be set to a specific angle, it instead uses that value to set the speed in which the gears rotate. It is possible to change the code to have fixed values but I personally don't want to change the app provided just incase I mess it up in some way. Also I am still a novice when it comes to coding.

TLDR: Get 180° servos, not the 360° servos.

piyushdotcomm, 2025-06-21:
360° servos will be better for full controlled movement. As 180° servos might have some limitations.